RobTillaart/ACS712 #13: A little confusion about how the Ampere works.

PanduHalimie31, 2022-07-05:
Hello, excuse me. I got some question to ask.
Im using 5A ACS712 Chip for my device. One thing i realized is without anything (any load) connected to my device. It's still showing up 30-45mA. I'm a little confused on why is this happening.
Another question is, when i charge my phone it shows the current is around 100-150mA. Meanwhile it should be around 2A/2000mA. Any idea on why is this happening?

Thank you in advance.

RobTillaart, 2022-07-05:
Thanks for your question. 
There is too little information to analyze your case, so I start with some generic questions.
What processor board do you use?
How is the ACS onnected?
Have you initialized the sensor to 5A?

A low current when nothing connected can be noise.
If you disconnect the device, what current does it report?

RobTillaart, 2022-07-05:
Q5 Have you compared your ACS712 measurements with a multimeter?

RobTillaart, 2022-07-11:
What voltage do you measure with DMM) between GND and VCC ?

PanduHalimie31, 2022-07-11:
Thanks you! 
So, yes. I do have a 1nF Caps on the filter pin. Here's are the schematic, its connected to a relay.

[image]

The voltage i got by measuring those pins are 2.534V (both GND - VCC and GND - VIOUT).
I called **ACS.autoMidPoint();** on my code already which show 511-512 since it ADC/2. Or it's different from what you mentioned above?

RobTillaart, 2022-07-11:
GND VCC  should be sround 5V, that cannot be 2.534V, please measure again

AutoMidpoint seems to return a value as expected.

PanduHalimie31, 2022-07-11:
Ah yes, my bad. It's 5V for GND - VCC pins. Lost focus for a moment.

RobTillaart, 2022-07-11:
Think I found the cause
From the datasheet

[image]

Please note that the ATMEGA 328 has 1024 levels ADC for 5V
The sensor only uses +- 1V for 5A so around 200 steps of the ADC are used for 5A.
That means that 1 step is around is about 25mA 

So the 30-60 mA you see is just one or two bits of the ADC, which is probably noise.

possible solutions:
- call **setNoisemV(70)** in setup - adjust the value so it just removes the noise.
- reduce the noise in hardware with a low pass filter => resistor 1K + capacitor (10 uF) between VIOUT and GND
- use an external ADC like the ASA1115 which has a higher resolution. Might still need a low pass filter.

RobTillaart, 2022-07-11:
The ADS1115 is a 16 bit ADC while the 328 has only 10 bit, that means 64 x precision (in theory).
That means 1V has 12800  steps so 1 step is in the order of 0.1 mV

You won't get all that extra accuracy in practice as such ADC is more sensitive for noise 

*in theory there is no difference between theory and practice, but in practice there is.*

PanduHalimie31, 2022-07-12:
Ah i see the issues now.
Will try the possible resolution soon and see what i got.
Thanks for time and of course the answer! Will inform you soon about the update of my device.

PanduHalimie31, 2022-07-13:
Tried out changing the NoisemV and the numbers im getting is much better and realistic than before. For now, the problems is fine at all no need to continue as im getting a better realistic numbers overall. Also i used setMidPoint(512) so the midpoint wont fluctuate.

Thanks for your time!

closed 2022-07-13 by RobTillaart

panbak, 2022-07-27:
Sorry in advance as the question is irrelevant to the ACS712 library. 
You mentioned to use a low-pass filter of 1K resistor + 10 uF capacitor. The filter will have a cut-off frequency of ~16 Hz.
How did you come up with the values of resistor and capacitor? I am asking as I need a low pass filter for my 20A ACS712 sensor.
Thank you and again excuse me for the irrelevant question.

RobTillaart, 2022-07-27:
@panbak 
Was just a starting point for a possible solution. Did not calculate them. 
In the datasheet page 11++ there is information about filtering.

panbak, 2022-07-27:
Thank you! Also I don't understand what the second parameter is when calling `ACS712  ACS(25, 5.0, 4095, 185);` Is it the sensor operating voltage?

RobTillaart, 2022-07-27:
**float volts = 5.0**  is the voltage of the internal ADC 
For an UNO and NANO this is typically 5.0 volts, for an  ESP32 this is 3.3V.  
This voltage is used to calculate the **_mVpstep** that is used with the **analogRead()** to measure the sensors output.

Might need explanation in the readme.md file

RobTillaart, 2022-07-27:
something like this 

## Interface

#### Base

- **ACS712(uint8_t analogPin, float volts = 5.0, uint16_t maxADC = 1023, uint8_t mVperA = 100)** constructor. 
It defaults a 20 A type sensor, which is defined by the default value of mVperA. See below.
Volts is the voltage used by the (Arduino) internal ADC.
maxADC is the maximum output of the internal ADC.
mVperA is defined by the sensor used.
These two values are used to calculate the voltage output of the ACS712 sensor.

mpictor, 2025-11-08:
> If I understand the working of your solution correctly the OPAMPs transform the (small) output range you are interested in to a larger output range. This effectively is a linear transformation (assumption), with both a factor (slope) and an offset (zero point). However there is a difference for DC and AC.

Exactly! It _seems_ linear enough for my purposes, but I haven't done enough to quantify it to be sure exactly how much distortion there is - and it gets more complicated with AC.

I did it this way simply because I could; I'd strongly recommend that people solve the issue with better ADCs rather than opamps.  14- and 16-bit ADCs are available for about $2 each from digikey, and I'm sure they are considerably cheaper elsewhere.