RobTillaart/ACS712 #14: Encountered a watt calculation problem after having two loads.

PanduHalimie31, 2022-07-13:
Hello, it's me again but with another problem.

I was measuring the device with a single load (22W Fan) and it works just fine, below is the measurements result:
[image]

Then i measured my second fan (20W Fan) and here's the result:
[image]

And here's when i used both fan as the loads.
[image]

Wierdly, the image show the watt is not working normally. I used Wattmeter device for comparison and it shows 40Watt but the ACS seems not working normally. Tried with other load too and there goes the same problem again.

Here's the code i used for calculating the Ampere and the Watts
[image]

RobTillaart, 2022-07-13:
Warning: never post an image of code as those are pretty hard to run on a computer (I cannot test it).

your code is wrong.
Debug it.

RobTillaart, 2022-07-13:
Oops wrong button.

RobTillaart, 2022-07-13:
if mA == 218  how can Watt become 17.58 ?
Follow the math you do with paper and pencil

PanduHalimie31, 2022-07-13:
Yes, there's some miscalculations on the codes.
But i fixed it already with the help of Arduino Forum by using 
`float Watt = ((220L*mA)/1000.0);` instead.

Thanks for answer and time sir!

closed 2022-07-13 by PanduHalimie31

RobTillaart, 2022-07-13:
What output do you get now?

The solution of the forum is not 100% optimal.
```cpp
float Watt = mA * (220 / 1000.0);
```
The compiler will optimize the division compile time so in it will in essence be
```cpp
float Watt = mA * 0.22;
```
As multiplications are faster than divisions on most micros, this solution will be slightly faster too.

The forum code will not be optimized and will have both a multiplication and division.
(for most projects the difference is not relevant, but for real time it may just make the difference)

----

Better use the forum for all code problems, there are far more people there to help out.
Only when there are problems in the library code itself you should open an issue under a library.