RobTillaart/ACS712 #15: problem with ESP32

rollins17, 2022-08-09:
Hello

with a Promicro 5v and : ACS712  ACS(A0, 5, 1023, 100);

I got :
No load:
analogRead(A0)    totalmA
516 -                    0.00

With load
551 -                    1.94
and 1.89 with multimeter, so it's OK

but with EPS32 and : ACS712  ACS(34, 3.3, 4095, 100);
No load:
analogRead(34)    totalmA
128 -                   0.00

With load
145 -                   0.16
here the code:
#include "ACS712.h"

// Arduino UNO has 5.0 volt with a max ADC value of 1023 steps
// ACS712 5A  uses 185 mV per A
// ACS712 20A uses 100 mV per A
// ACS712 30A uses  66 mV per A
/*
Promicro 5V
A0    totalmA
516 - 0.00
551 - 1.94

esp32 3.3V
34  totalmA
128 - 0.00
145 - 0.16

 * 
 */
//ACS712  ACS(A0, 5, 1023, 100);
// ESP 32 example (requires resistors to step down the logic voltage)

with 2x4.7k voltage divider.
 ACS712  ACS(34, 3.3, 4095, 100);

float lastValue = 0;
int nbrsample = 10;

void setup() {

  Serial.begin(115200);
  Serial.println(__FILE__);
  ACS.autoMidPoint();
  

}

void loop() {
  float  totalmA = 0;
  
  for(int nbr =0; nbr < nbrsample; ++nbr){
    totalmA += ACS.mA_DC();
    delay(10);
  }
  totalmA = (totalmA / nbrsample)/1000;

  Serial.print(analogRead(34));Serial.print(" - ");Serial.println(totalmA);

  delay(200);
}
If you could help me, it will be great because I don't understand.
Best regards.
Laurent

RobTillaart, 2022-08-09:
Thanks for creating the issue. 
As the ACS712 and library works as it should ( = verified by the Promicro tests ) the problem might be in the ADC of the ESP32 or the voltage divider.

#### Analysis
- using a voltage divider should give a voltage of around 1.25V 
Can you confirm this with a multimeter?
- A voltage of 1.25V should give an **analogRead()** of 1550, definitely not 128
- A current of 1.75A  and a setting of 100mV/Amp (last parameter) should give an offset voltage of 175 mV from the ACS.
Can you measure this with a multimeter?
- An offset voltage of 175 mV should be 175 + 1250 = 1425 mV = 1425/3300*4095 = 1768 (approx).
- my prime suspect is the voltage divider, 
Can you measure the resistance of both resistors?

RobTillaart, 2022-08-09:
Found this link - https://randomnerdtutorials.com/esp32-adc-analog-read-arduino-ide/
As we are not near the edges of the range it is not the cause, still an interesting read.

rollins17, 2022-08-09:
Dear Rob

You're right, I have 1/8w resistors and I made confusion with red and orange, I have changed with 2 10 k, check with potentiometer = 0 to 4095.

Here the results;
No load-----------
1228 - 0.03
1227 - 0.06
1227 - 0.04
1223 - 0.02
1219 - 0.04
1217 - 0.01
load----------------
1431 - 0.62
1326 - 0.90
1322 - 0.88
1328 - 0.85
1328 - 0.87
1311 - 0.87
1328 - 0.85
on multimeter 1.66A 
ACS at 3.3, 4095

It's seem to be the half.
Any ideas ??

Regards 
Laurent

RobTillaart, 2022-08-09:
Ok, we're closing in ...

RobTillaart, 2022-08-09:
The factor two is caused by the voltage divider.
It will also divide the offset by two.

Might be an interesting setting for the library

rollins17, 2022-08-09:
Great Thanks Rob

Have a nice day

Laurent

RobTillaart, 2022-08-09:
@rollins17 
If issue is solved you may close the issue.

rollins17, 2022-08-09:
Sorry

RobTillaart, 2022-08-09:
@rollins17 
I will create a 0.3.0 version with support for the voltageDividerFactor.
Are you willing to test / verify it?
Expect a new branch later today.

rollins17, 2022-08-09:
Hi Rob

Of Course !!

an other thing it's for my solar panel, the esp can reset at anytime but the panel will  be on line, I understood the the MidPoint  function is a calibrate point, for accurate measurements,  can we skip  this function and fix it in program  ?

Regards‌
Laurent
 

De : "Rob Tillaart"
A : "RobTillaart/ACS712" ,"rollins17" ,"Mention"
Envoyé: mardi 9 Août 2022 19:48
Objet : Re: [RobTillaart/ACS712] problem with ESP32 (Issue #15)
 

 

@rollins17
I will create a 0.3.0 version with support for the voltageDividerFactor.
Are you willing to test / verify it?
Expect a new branch later today.

RobTillaart, 2022-08-09:
> an other thing it's for my solar panel, the esp can reset at anytime but the panel will  be on line, 
I understood the the MidPoint  function is a calibrate point, for accurate measurements,  
can we skip  this function and fix it in program  ?

Could work without if there was no voltage divider. I assume the midpoint is shifted too.

RobTillaart, 2022-08-09:
Branch is created - https://github.com/RobTillaart/ACS712/tree/develop

rollins17, 2022-08-09:
Thanks

tomorrow morning  I will test it

Regards

RobTillaart, 2022-08-10:
Had some insights tonight.
If you would change the parameters in the constructor to. ACS(pin, 1.65, 4095, 50). It should work with the voltage divider too.

Can you confirm?

If so the new function is not needed, and could be replaced by a runtime configurable voltage etc instead of a construct time only setting.  That would be a better design. 

To be continued.

rollins17, 2022-08-10:
Hello Rob
With the 02.6 library :
ACS(pin, 1.65, 4095, 50) give 
ACS(pin, 3.3, 4095, 50) give 1.66 correct

Regards
Laurent.

rollins17, 2022-08-10:
ACS(pin, 1.65, 4095, 50) give 0.91

RobTillaart, 2022-08-10:
Thanks for testing, 

OK, my nightly thoughts were in the right direction, but apparently not 100%  😁

Need to rethink the interface, 
- mV/A  should be a float so it can be set more exact e.g. 49.6 instead of 50 would be possible then. 

The current 0.3.0 development branch will become obsolete as the problem was solvable in 0.2.6.

RobTillaart, 2022-08-10:
A new develop branch - 0.2.7 - is created to support a float mV/Amp. 
It will include the "voltage divider" insights of this thread.

RobTillaart, 2022-08-10:
@rollins17 
develop branch - 0.2.7 - is stable, if you have time, can you test it works for you?

rollins17, 2022-08-10:
Yes of course

RobTillaart, 2022-08-12:
@rollins17 
As no problems were found I merged it into 0.2.7

closed 2022-08-12 by RobTillaart