RobTillaart/ACS712 #23: Investigate noise

RobTillaart, 2022-08-28:
See https://github.com/RobTillaart/ACS712/issues/18 for background info

## Problem description
The class sees "ghost current" when there is none. Causes known:
- **mA_AC()** will interpret the noise band as min-max and multiplies that by the formfactor.
- **mA_AC_sampling()** will add up noise when there is zero current.

The class has a **_noisemV** that is configurable, but the measurements do not use this for noise - detection or reduction.
(however it is used in **mA_AC()** for zero detection)

## Analysis
### Current < noise level  (including zero)
If samples below the noise level are skipped, 
- **mA_AC_sampling()** will return 0 as there are no samples to add.
- **mA_AC()** will not be affected as it uses min and max which are the same => result = 0 mA.

If samples below the noise level are NOT skipped
- **mA_AC_sampling()** will add them  which might result in a too large current.
- **mA_AC()** will have a min and max that are probably larger than the signal resulting in too large current. 

So should small values be regarded as zero?

### Current > noise level
If samples below the noise level are skipped, 
- **mA_AC_sampling()** will miss some of the small values and result might be too low.
- **mA_AC()** will probably not affected as it only looks for min and max which will be outside the noise level

If samples below the noise level are NOT skipped
- **mA_AC_sampling()** will add noise on the signal as signal causing deviation. 
Some of these values will be too high and some too low. netto effect is hard to predict.
- **mA_AC()** can have a min and max that are both farther from midPoint as noise is added/subtracted. 
Effectively the noise level might be added  twice, resulting in a too large current. however 

## Thoughts / actions
- are there electronic means to reduce the noise? (capacitors etc of course preferred )
- how large are the deviations?
- can we detect noiseLevel automatically, e.g. together with the autoMidPoint?
- ...

RobTillaart, 2022-08-28:
@imk2021 
created an issue for the noise with some thoughts...

imk2021, 2022-08-31:
Hello Rob and many thanks for the kind words re old electronic mates, 
yes we had a fine time.
One ran his own electronics business for a life time, other end up 
teaching at local university, while took the IBM road doing large IT 
systems for Airlines.
Also many thanks for the low pass filter info, i have lots of electronic 
bits so will put something together in next couple of days.
But tomorrow I have to get on with mundane  shelving job in my 
office/man cave as this desk clutter is getting out of control :-)

Now I don't know what to say about latest test result other than MEGA 
JOB ROB!

I tested as in test 4 but added mixed/very complex sine plus dimmed sine 
and well see results for yourself FIVE STAR and big smile on my face.
Result and images attached below

So Rob have a good evening and well deserved rest Ian.

Test Set Five 31-Aug-2022 2100
mA_AC_sampling() 40 calls per sec.
OVEN_CONTROLLER_PRODUCTION_V5
Development library version 0.3.0

ACS712 Calibration load 1000w Brymen 4548 mA

ACS712  ACS( ACS712_CURRENT_SENSOR_ANALOG_PIN , 5.0, 1023, 101 );        
MeanmA=4473
ACS712  ACS( ACS712_CURRENT_SENSOR_ANALOG_PIN , 5.0, 1023, 100 );        
MeanmA=4515
ACS712  ACS( ACS712_CURRENT_SENSOR_ANALOG_PIN , 5.0, 1023, 99 );        
MeanmA=4576
ACS712  ACS( ACS712_CURRENT_SENSOR_ANALOG_PIN , 5.0, 1023, 98 );        
MeanmA=4630
ACS712  ACS( ACS712_CURRENT_SENSOR_ANALOG_PIN , 5.0, 1023, 97 );        
MeanmA=4675

Therefore all test result below where at  ACS712 Calibration value of 99

Other Load Testing Below:

Not Dimmed
100w  Brymen=0.429  MeanmA=432
200w  Brymen=0.832  MeanmA=838
300w  Brymen=1.257  MeanmA=1266
500w  Brymen=2.064  MeanmA=2082
1000w Brymen=2.562  MeanmA=2607

200w Dimmed
Brymen=0.508    MeanmA=509 Min Dim
Brymen=0.602    MeanmA=599
Brymen=0.700    MeanmA=698
Brymen=0.828    MeanmA=830 Max Dim

500w Dimmed
Brymen=1.274   MeanmA=1270 Min  Dim
Brymen=1.301   MeanmA=1299
Brymen=1.403   MeanmA=1397
Brymen=1.500   MeanmA=1499
Brymen=1.601   MeanmA=1599
Brymen=1.704   MeanmA=1712
Brymen=1.804   MeanmA=1803
Brymen=1.900   MeanmA=1903
Brymen=2.005   MeanmA=2014
Brymen=2.054   MeanmA=2062 Max  Dim

100w Sine 200w Dimmed
Brymen=0.779   MeanmA=787 Min  Dim     Image 100wPlus200w-Dim-1.jpg
Brymen=0.906   MeanmA=914        Image 100wPlus200w-Dim-2.jpg
Brymen=1.001   MeanmA=1008        Image 100wPlus200w-Dim-3.jpg
Brymen=1.101   MeanmA=1108        Image 100wPlus200w-Dim-4.jpg
Brymen=1.203   MeanmA=1208        Image 100wPlus200w-Dim-5.jpg
Brymen=1.252   MeanmA=1263 Max  Dim    Image 100wPlus200w-Dim-6.jpg

RobTillaart, 2022-09-01:
@imk2021 
The results are better than expected. Calibrating pays of very well.
You can configure floats like 99.5 mVperAmpere if you want to improve on this 

The photo's didn't make it to Github unfortunately, but the numbers seem all to be within 1% (at first sight)

I'm going to merge now. 

----

Q: in your opinion, do you need both the **mA_AC_samples()** and the **mA_AC()**?
The **mA_AC_samples()** is able to measure all signal types. No need for the from factor. 
Would make the footprint of the library a bit smaller without loosing functionality.

imk2021, 2022-09-01:
Rob

I added the images inline in this post as think it is important to see 
how the combined sine and dimmed looked.

Ian

RobTillaart, 2022-09-01:
> Rob I added the images inline in this post as think it is important to see how the combined sine and dimmed looked. Ian

Can't see them...

RobTillaart, 2022-09-02:
@imk2021 
Do we need additional functions for noise analysis?
I think the **mA_peak2peak()** with zero (or constant DC) current gives all information needed to determine the noiselevel.

A wrapper function can be made to determine the noiseLevel in mV
```cpp
float mVNoiseLevel(float frequency, uint16_t cycles)
{
   float mA = mA_peak2peak(frequency, cycles);
   return mA * _mVperAmpere * 0.001 / 2;   //  divide by 2 as the level is half of the range
}

...

ACS.setNoiseLevel(mVNoiseLevel(50, 1));
```

Opinion?

imk2021, 2022-09-03:
Hello Rob,

Sorry not got back to you, but still busy doing domestic stuff.
Think your proposed "float mVNoiseLevel(float frequency, uint16_t 
cycles)"      Will be fine.

My next testing session will included the RobotDyn dimmer I did original 
tests with, to see how well system performs with all the extra 
timer/interrupts etc.
It is not a deal breaker for me if it doesn't work too well as I can off 
load the RobotDyn dimmer function to a second NANO and control it via 
serial command.
Get back to you in couple of days, but have more shelving to do in man 
cave :-)

Good weekend Ian

RobTillaart, 2022-09-03:
Thanks for the info,  I think most gain with the noise level will be in s low pass filter. Preferably one with a potmeter to adjust the frequency.

Think one need less averaging then.

The dimmer uses hardware interrupts and "affects" the measuring loop for the mA. A second controller is the best way to 'separate concerns '. 

Success with "reorganizing" your man cave, I always find stuff that distracts me.

RobTillaart, 2022-09-03:
#### Thought: 
Noise does not need to be symmetrical, or is it always? or is the difference neglectable?
Noise above the midpoint could be larger than noise under the midpoint. 
No plan to investigate this.

RobTillaart, 2022-09-03:
A new develop branch is made with the new **mVNoiseLevel()** function.  To be continued later next week.
Have a good weekend!

RobTillaart, 2022-09-03:
Idea for the LOW PASS filter

Capacitor 1µF and 2 potmeters in series, one of 2500 Ω (coarse) and one of 100 Ω  (fine)
Don't know if that would improve quality of measurement?

A digital potmeter would be interesting as the Arduino could calculate the Cut Off frequency, display it etc.

RobTillaart, 2022-09-04:
Did a quick test today with an ACS712 I had found.  Put my BRYMEN in series with ACS72.
I used a halogen lamp with an unknown wattage.

First I got strange measurements  - numbers are from my head
```
BRYMEN 128 mA
ACS712 235 mA   (mA_AC_sampling)
ACS712 230 mA   (mA_AC)
```
mmmm, cannot be right. 
I checked the ratio ACS/ BRYMEN which was about 1.85 and that triggered to recompile for a 5A ACS712.
This uses 185 mV/Amp while the 20 A version uses 100 mV/Amp.

After a quick compile I got far better results:
```
BRYMEN 129 mA
ACS712 131 mA   (mA_AC_sampling)
ACS712 127 mA   (mA_AC)
```

So the 5A sensor is also confirmed working within the 1-2% accuracy range.

I added the code to the examples, however the I2C_LCD library I used is not supported in the build-CI .
Therefor I had to exclude it from compilation. Still the code can be useful as a starter.

imk2021, 2022-09-04:
Hello Rob

Well done with the ACS712 testing at 5A another goodly milestone for 
your ACS712 library :-)
BTW what sort of numbers do you get for zero load i typically get about 
25mA but sometimes as high  60mA at ADC input of 511/512 .

Tonight I hope to do quick test with the RobotDyn dimmer included just 
to see.
If result look promising I'll do the full range of tests, will let you know.

Shelving project still in progress, hope to complete by next weekend.

Ian

RobTillaart, 2022-09-04:
Zero load measurement
mA_AC() = 7.0
mA_AC_sampling() = 44.6

[image]

30 Watt Halogen lamp
mA_AC() = 107.7  (strongly fluctuating)
mA_AC_sampling() = 130.9   (far less fluctuating) - Matches BRYMEN 129 mA. (note not calibrated)

[image]

RobTillaart, 2022-09-04:
What I learned during my latest test is 
1. that the 5V which the ACS gets from Arduino is not stable enough.
2. that one might need to call **ACS.autoMidPoint();** e.g. once per X minutes.

On my list is a test with 3V3 to power the ACS. The Arduino 3V3 is far more stable.

RobTillaart, 2022-09-05:
### DIMMER TEST

#### no dimmer

100W Load Not Dimmed
Brymen 0.432A         MeanmA from 40 X mA_AC_samples() = 436 (As expected result)

#### dimmer 0%

With RobotDyn code included:
100W Load Not Dimmed
200W on RobotDyn  Dimmer setPower( 0 )        (It is interesting that at ZERO Power the RobotDyn library still drives the dimmer board, see image and power output)
Brymen 0.470A         MeanmA from 40 X mA_AC_samples() = 503

[image]

#### dimmer 25%

100W Load Not Dimmed
200W on RobotDyn  Dimmer setPower( 25 )
Brymen 0.752A         MeanmA from 40 X mA_AC_samples() = 942

[image]

#### dimmer 50%

100W Load Not Dimmed
200W on RobotDyn  Dimmer setPower( 50 )
Brymen 01.066A         MeanmA from 40 X mA_AC_samples() = 1277

[image]

RobTillaart, 2022-09-05:
What is clear from the scope signals with **dimmer** is that that device substantially adds to the noise.
(compare with images above).

The software of the library does just apply its math to this raw data including every noisy spike.
As the software cannot know in advance how the signal should look like, it is hard or even impossible to correct these spikes.

I will try to adjust the **mA_AC_sampling()** version so it will dampen the spikes a bit.

RobTillaart, 2022-09-05:
updated develop branch
- add **suppressNoise(bool flag)**

The noise suppression will average two samples before doing the math. 
- This should dampen the spikes
- math has 50% less samples to work with (simulation in excel looks OK).

RobTillaart, 2022-09-08:
(Ian's test results 2022-09-08)
Test Set Seven   8-Sept-2022 1130
mA_AC_sampling() 40 calls per sec.
OVEN_CONTROLLER_PRODUCTION_V5
Development library version 0.3.1

Room Light Dimmer,  suppressNoise() testing

suppressNoise( false )
100W Load Not Dimmed
Brymen=0.418          MeanmA from 40 X mA_AC_samples() = 423  

100w Sine 200w Dimmed (~50% sine)
Brymen=1.089   MeanmA=1098

suppressNoise( true )
100W Load Not Dimmed
Brymen=0.418          MeanmA from 40 X mA_AC_samples() = 424  

100w Sine 200w Dimmed (~50% sine)
Brymen=1.091   MeanmA=1096

RobTillaart, 2022-09-16:
@imk2021 

Any progress with the hardware filtering?

RobTillaart, 2022-10-10:
FYI, released 0.3.1 version.

imk2021, 2022-10-11:
Hello Rob,

Really sorry but not had chance to do any filter testing as still 
sorting man cave stuff.
However, my guess is a lot of the noise i am seeing is due to the test rig.
This being wired up on a breadboard with jumper wire which are just lots 
of small aerials.
Hence the noise from multiple switching power supplies in PC, Big dual 
monitor and the Hantek scope screen all seems to be added in.

Going to put rig on matrix pad board and short wires with proper 
grounding then will get best filter result testing I think.

How is all with you and your testing going?

Ian

RobTillaart, 2022-10-11:
Doing not too bad, thinking about energy conservation & reduction more than I used to.

Only a few activities on the ACS712 library, reviewed the code and some minor edits, and released it yesterday.

Most time was spent on other libraries, wrote a **FRAM ringbuffer** class to store measurements (whatever) persistent in FRAM. This allows to resume processing after a reboot. It is work in progress. Another new library is for the **CHT8305** temperature and humidity sensor, it has a build-in **ALERT** function if temperature and/or humidity is going too high. That is far more efficient than a constant polling of the sensor. Hope to get the hardware this week.

RobTillaart, 2022-11-21:
@imk2021 
I released 0.3.2 as 0.3.1 got a bug for the DC measurements.

in the meantime, did you made any progress or should we close this issue?
(I did not make any progress on the noise preventing front)

RobTillaart, 2023-01-17:
@imk2021 
Released 0.3.4 yesterday, new feature is external ADC support (up to 16 bit), which might allow new ways to less noise.
As I have to reschedule my time I close this issue for now - It fell of my prio list, sorry ;)
Feel free to reopen if new ideas or insights pop up!

closed 2023-01-17 by RobTillaart