RobTillaart/ACS712 #24: Testing version 0.3.0

RobTillaart, 2022-08-31:
Hello Rob,

Sorry something wrong with 0.3.0 as mA_AC_sampling () returns a large 
number eg 319445 mA when connected to the ACS712 at zero current.

Test Rig changes to verify results are:
Hardware, Connected A0 to A1.

Software, Added lines of code to display ADC output of A1 and raw valve 
at A0
MaxmA = analogRead( A1 );
lcd.setCursor(0, 0);             lcd.print( "Max mA's: " ); lcd.print( 
MaxmA );

Test a.
ACS712 connected to A0 and A1 (As normal)
Output A1=513, mA_AC_sampling () output= 319445 mA

Test b.
Disconnected ACS712 form A0 and A1
Connected 10k Variable resistor between 0v and vdd (5v rail) and 
adjusted to 1/2 vdd
Connected resistor wiper to A0 and A1  (A1 Reading 511, mA_AC_sampling 
() returns 22.62mA)
---------------- This test has no ACS712 noise ---------------------

Test c
wiper of 10k Variable resistor still connected to A0 and A1
But connected ACS712 to A0 and A1 via a 10uF tantalum capacitor, thus 
injecting ACS712 noise into A0 and A1
A1 Reading is still 511, but mA_AC_sampling () output is a large 
changing values in the region of 255234  mA

Test d.
Disconnecting ACS712 from the 10uf capacitor and values are A1=511, 
mA_AC_sampling () returns 26.12mA' ish)

It seems that 0.3.0 is super sensitive to ACS712 noise.

Sorry to be bring of bad news Rob, have a good evening, Ian

_Originally posted by @imk2021 in https://github.com/RobTillaart/ACS712/issues/22#issuecomment-1232144590_

RobTillaart, 2022-08-31:
@imk2021 
Moved the discussion into a new issue .

RobTillaart, 2022-08-31:
## Analysis tests
### Test a.
> Output A1=513, mA_AC_sampling () output= 319445 mA

Cannot explain.
Do you see this problem in both the 0.2.8 branch and the 0.3.0 branch ?  
If not it would indicate a problem in my merging. 

So please compare the 0.2.8 and 0.3.0 libraries  - mA_AC_sampling().
- with zero load
- with 1000 W load => does the problem also occur with serious load?

Can you measure the mA_peak2peak() with no load?

### Test b.
> A1 Reading 511, mA_AC_sampling() returns 22.62mA

This is expected as the AC_sampling() samples the noise. 

### Test c.
> A1 Reading 511, mA_AC_sampling () returns 255234  mA

Identical (not understood) cause as test a. 

### Test d.
> A1=511,mA_AC_sampling () returns 26.12mA

As expected.

## Noise

> It seems that 0.3.0 is super sensitive to ACS712 noise.

I would turn that around, the ACS712 introduces a lot of noise.
1. from itself (the breakout?)
2. indirect from power supply

So
How do you power the ACS712?
From Arduino?
Separate power supply? preferred!
Can measure the stability of power

---- 

I must find time to setup a test bench and do some tests myself.

imk2021, 2022-08-31:
Morning Rob
Notes receive and i'll do testing for you later, however I am out today 
with old electronic eng. mates i went to college with 50 years ago.

Current rig is uno and it powers the ACS712 via 5v usb power from my 
quality power supply PC

But one thing I noticed last night as I put PC into sleep mode and 
uno/ACS712 system now running on PC USB standby power.
Was that the very large number coming out of  mA_AC_sampling() when 
ACS712 was connect went to low level of 40mA or so.
Also when I started this PC up this morning as it booted the numbers 
from mA_AC_sampling() varied significantly as the PC boot load changed.
Also when i put a scope prob on A0/A1 mA_AC_sampling() show very large 
numbers.

These thing didn't happen with 0.2.8 as I captured images with the scope 
for you, hence some change has made mA_AC_samplig() super sensitive to 
noise.

So it seems there is some tipping point in the system where minor noise 
spikes are correctly ignored but large spikes are seen as large sine 
currents. Maybe

When i get back home I will have close look at my rig for possible 
grounding issues and look into moving it.
 From a potentially noisy Hobby Board with Patch cables to a soldered up 
pcb like board ready for production.
As given the results we have with 0.2.8 I feel really confident that 
your ACS712 solution is very good and more than fit for what I want to do.

One other test I am going to do on 0.2.8 is a 200W switched/dimmed sine 
on top of a static 100w none dimmed sine.

Have a good day Rob and I'll get back to you with more results/info on 
my return. Ian

RobTillaart, 2022-08-31:
@imk2021

Think I found the cause, I reverted _midPoint to be signed int again. 
In the code the ```current = analogRed() - midpoint```  failed due to signed unsigned mixing. 

I have updated the develop branch with the fix.
So please download latest version before testing.

imk2021, 2022-08-31:
Evening Rob,

Just got home and given the updated  0.3.0 a quick test at 100w, values 
look good and the very large values have returned to normal.
I shall re do the 1000w calibration and do full test this evening, plus 
a dimmed sine on top of a non dimmed sine to test very complex  signals.

I am optimistic, have a good evening Ian

RobTillaart, 2022-08-31:
Good evening,

Did you have a nice day with your old mates? 
Think there were beautiful stories that brought back lots of memories. 
A day to remember!

----
FYI, how I did the analysis.

I had replaced the **analogRead()** calls with a simulation so I could quickly test different signals. 
That showed me where I went wrong so I could revert the datatype of the midPoint (reference level).

It also showed me a way to implement support for an external ADC like the MCP320X series.
That is an idea for the future.