RobTillaart/ACS712 #27: How can we measure exact AC current in Amps

atanudasgupta, 2022-11-18:
Hi,

I went over your source code and readme, I am using ACS712 with AC50Hz, 15A , with ESP32,  how can I measure amount of current flowing in the circuit

RobTillaart, 2022-11-18:
check - https://create.arduino.cc/projecthub/instrumentation-system/acs712-current-sensor-87b4a6

it is not with my library but explains how it is done,

RobTillaart, 2022-11-18:
Note, it is never exact, but the test results so far are pretty close to a professional Brymen multimeter.

PS, there is a bug fix pending, so either use the develop branch or the 0.3.0 version.
(will be merged later today)

RobTillaart, 2022-11-18:
If there is no current you should get 0 plus maybe some noise (see datasheet). 
One way to test that is to disconnect the current wires, so the ADC should measure the midpoint (around 0).
You can calibrate then by calling autoMidPoint().

If that does not work, your sensor may be broken.
As I dont know your hardware setup I can only guess.
As this seems not a library issue you better ask help on the Arduino forum.

RobTillaart, 2022-11-21:
FYI, 
I'm merging the bug fix within an hour or so into a new release 0.3.2

RobTillaart, 2022-11-21:
As this issue is not a library issue I close it. Feel free to reopen it if the problem is in the library.

closed 2022-11-21 by RobTillaart