RobTillaart/ACS712 #35: Non-blocking DC automidpoint?

astorun, 2023-04-19:
Hi,

Is it possible to add an option to make automidpoint non-blocking?

I am developing a firmware that involves touchscreen and during boot it adds a 5 second delay until display becomes functional since automidpoint  blocks the code execution. i adjusted automidpoint to cycle around 50 times to get a precise measurements.

I tried other methods like putting automidpoint in separate function and call it from loop but it creates various weird behavior in other parts of the program.

Thanks.

RobTillaart, 2023-04-19:
Thanks for the question.
The short answer is no, when measuring AC one needs to average a whole sine wave to determine the midpoint. 
The auto midpoint does this in a blocking way. 

What you might do is to do it once, get the value and store it in EEPROM. Next time you restart you just fetch it from EEPROM and set it. This might work, not tested. 

Another option is to write a function that samples once and returns an estimated midpoint and a confidence level. It would allow to spread the determination of the midpoint over time.

An idea how this function could work is tracking min snd max values of the samples.
If hese do not change confidence increases by 1%. If one changes confidence is divided by 2. 
If the confidence becomes 100% the midpoint could be the average of min and max. 

The latter uses random samples and the idea is that   they are made uniform spread over the sine wave so quite a chance you find the peaks and thus the average.
This idea is not tested either.

Does one of these ideas fit in your project/design?

RobTillaart, 2023-04-19:
The latter idea seems most interesting so I wrote a prototype to get you started

```cpp
//    FILE: estimateMidPoint.ino
//  AUTHOR: Rob Tillaart
// PURPOSE: experimental 
//     URL: https://github.com/RobTillaart/ACS712

#include "Arduino.h"

uint32_t start, stop;
int _pin = A0;
uint32_t count = 0;
volatile uint16_t mp;
float conf = 0;

uint16_t estimateMidPoint(float &confidence, bool reset = false)
{
  static bool _firstCall = true;
  static float _minimum, _maximum, _confidence;

  int value = analogRead(_pin);
  if (_firstCall || reset)
  {
    _firstCall = false;
    _minimum = _maximum = value;
    _confidence = 0;
    confidence = _confidence;
    return _minimum;
  }

  if (value > _maximum)
  {
    _maximum = value;
    _confidence /= 2;
  }
  else if (value < _minimum)
  {
    _minimum = value;
    _confidence /= 2;
  }
  else if (_confidence < 100)
  {
    _confidence += 1;
  }

  confidence = _confidence;
  return (_minimum + _maximum) / 2;
}

void setup()
{
  Serial.begin(115200);
  while (!Serial);
  Serial.println(__FILE__);
}

void loop()
{
  count++;
  start = micros();
  mp = estimateMidPoint(conf, true);
  stop = micros();
  Serial.print(millis());
  Serial.print("\t");
  Serial.print(count);
  Serial.print("\t");
  Serial.print(conf);
  Serial.print("\t");
  Serial.print(mp);
  Serial.print("\t");
  Serial.print(stop - start);
  Serial.println();
  delay(random(100));
}

//  -- END OF FILE --

```

### output
```
estimateMidPoint.ino
2	1	0.00	709	216
10	2	1.00	709	160
60	3	2.00	709	156
134	4	3.00	709	156
193	5	4.00	709	160
224	6	5.00	709	160
297	7	6.00	709	160
343	8	7.00	709	156
421	9	8.00	709	156
446	10	9.00	709	160
456	11	4.50	709	148   <<<<<<<<<<
497	12	2.25	709	156   <<<<<<<<<<
564	13	3.25	709	152
657	14	4.25	709	164
700	15	5.25	709	160

...

5123	108	98.25	709	164
5221	109	99.25	709	156
5316	110	100.25	709	160
5321	111	100.25	709	148
5331	112	100.25	709	152
```

Not tested with an ACS712, just with a floating analog and the first runs look not bad.

First call (UNO) takes 216 usec. remaining ~164usec.
(not to bad compared to 20 millis blocking)

RobTillaart, 2023-04-19:
Note: the confidence increases just by one now, but you could make steps of 2,3,5, 8,13, ... so it converges faster.
(Fibonacci always looks cool in algorithms)
Maybe as parameter to the function, default 1.

```cpp
uint16_t estimateMidPoint(float &confidence, int step = 1,  bool reset = false)
```

RobTillaart, 2023-04-19:
Note2: please note that the **estimateMidPoint()** does not set the midPoint yet.

RobTillaart, 2023-04-19:
> Sounds good. That might work in our application. Thanks!

Please let me know how well it works, I might add the function to the library. 
And tested functions make most sense 😎

RobTillaart, 2023-04-19:
@astorun 

If you want to measure the zero level of DC (as stated in the title), which has no frequency, you might just call 
```cpp
value = autoMidPoint(1000, 1);
```

By setting the freq very high the two periods it blocks become much shorter. In the example above 1000 Hz will block 2 ms.
That will give a reasonable amount of samples to average,

But maybe there should be an **autoMidPointDC(uint16_t cycles = 8)**, just taking 8 samples and average them.

```cpp
uint16_t ACS712::autoMidPointDC(uint16_t cycles = 8)
{
  if (cycles == 0) cycles = 8;
  uint32_t total = 0;
  for (uint16_t i = 0; i < cycles; i++)
  {
    total += _analogRead(_pin);
    subTotal += reading;
  }
  _midPoint = (total + (cycles/2))/ cycles;    // rounding.
  return _midPoint;
}
```

Assumption is that there is no current flowing...

astorun, 2023-04-19:
I integrated your code to to acs712.cpp and made some fixes:
```
uint16_t ACS712::autoMidPointDC(uint16_t cycles)
{
    if (cycles == 0) cycles = 1;
    uint32_t total = 0;
    uint32_t subTotal = 0;
    uint32_t reading  = 0;
    for (uint16_t i = 0; i < cycles; i++)
    {
      total += analogRead(_pin);
      subTotal += reading;
    }
    _midPoint = (total + (cycles/2))/ cycles;    // rounding.
    return _midPoint;
}
```

added the following line to acs712.h
`uint16_t autoMidPointDC(uint16_t cycles = 1);`

and calling it within setup like this:
`ACS.autoMidPointDC(100);`

Works like a charm! No more blocking.

I suggest you to add this to library!

RobTillaart, 2023-04-19:
will add the function **autoMidPointDC()** asap.

Do you have an opinion about the **uint16_t estimateMidPoint(float &confidence, bool reset = false)** which should work for the AC current - midPoint determination? (no is a perfect OK answer)

RobTillaart, 2023-04-19:
@astorun 

Your proposed code adds **reading** to **subtotal** which is never used.  **reading** is always zero BTW.

Q: on which platform did you test it?

RobTillaart, 2023-04-19:
@astorun 
Created develop branch and a Pull Request - https://github.com/RobTillaart/ACS712/pull/36

closed 2023-04-20 by RobTillaart