RobTillaart/ACS712 #40: Build Errors with ESPHome 2023.7.1

chris-kuhr, 2023-08-09:
Hi,

I am running ESPHome in Docker on an RPi4B 2G.
My other devices work just fine on the Olimex ESP32-PoE-ISO. 
I saw the erros shwon below in my installation and updated the docker image just an hour ago, but the errors persist.

Here are the logs:

```INFO ESPHome 2023.7.1
INFO Reading configuration /config/energy-probe.yaml...
WARNING GPIO12 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Compiling app...
Processing energy-probe-appliance (board: esp32-poe-iso; framework: arduino; platform: platformio/espressif32@5.3.0)
--------------------------------------------------------------------------------
Removing unused dependencies...
Library Manager: Installing git+https://github.com/RobTillaart/ACS712.git
INFO Installing git+https://github.com/RobTillaart/ACS712.git
git version 2.30.2
Cloning into '/config/.esphome/platformio/cache/tmp/pkg-installing-ukh35_rd'...
Library Manager: ACS712@0.3.7+sha.c56d3ce has been installed!
INFO ACS712@0.3.7+sha.c56d3ce has been installed!
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- ACS712 @ 0.3.7+sha.c56d3ce
|-- WiFi @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- Update @ 2.0.0
|-- noise-c @ 0.1.4
Compiling .pioenvs/energy-probe-appliance/src/esphome/components/adc/adc_sensor.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/components/api/api_connection.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/components/api/api_frame_helper.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/components/api/api_pb2.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/components/api/api_pb2_service.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/components/api/api_server.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/components/api/list_entities.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/components/api/proto.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/components/api/subscribe_state.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/components/api/user_services.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/components/custom/sensor/custom_sensor.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/components/esp32/core.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/components/esp32/gpio.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/components/esp32/preferences.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/components/ethernet/esp_eth_phy_jl1101.c.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/components/ethernet/ethernet_component.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/components/md5/md5.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/components/mdns/mdns_component.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/components/mdns/mdns_esp32.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/components/mdns/mdns_esp8266.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/components/mdns/mdns_host.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/components/mdns/mdns_rp2040.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/components/network/util.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/components/ota/ota_backend_arduino_esp32.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/components/ota/ota_backend_arduino_esp8266.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/components/ota/ota_backend_arduino_rp2040.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/components/ota/ota_backend_esp_idf.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/components/ota/ota_component.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/components/sensor/automation.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/components/sensor/filter.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/components/sensor/sensor.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/components/socket/bsd_sockets_impl.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/components/socket/lwip_raw_tcp_impl.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/components/socket/socket.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/core/application.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/core/color.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/core/component.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/core/component_iterator.cpp.o
[... 67 lines cut ...]
 void delay(uint32_t);
      ^~~~~
In file included from src/esphome/components/adc/adc_sensor.h:4,
                 from src/esphome.h:3,
                 from src/main.cpp:3:
src/esphome/core/hal.h:39:6: note: previous declaration with 'C++' linkage
 void delay(uint32_t ms);
      ^~~~~
src/main.cpp:24:30: error: conflicting declaration of 'void esphome::delayMicroseconds(uint32_t)' with 'C' linkage
 #define delayMicroseconds(x) esphome::delayMicroseconds(x)
                              ^~~~~~~
/config/.esphome/platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal.h:133:6: note: in expansion of macro 'delayMicroseconds'
 void delayMicroseconds(uint32_t us);
      ^~~~~~~~~~~~~~~~~
In file included from src/esphome/components/adc/adc_sensor.h:4,
                 from src/esphome.h:3,
                 from src/main.cpp:3:
src/esphome/core/hal.h:40:6: note: previous declaration with 'C++' linkage
 void delayMicroseconds(uint32_t us);  // NOLINT(readability-identifier-naming)
      ^~~~~~~~~~~~~~~~~
Compiling .pioenvs/energy-probe-appliance/lib18f/WiFi/WiFiClient.cpp.o
Compiling .pioenvs/energy-probe-appliance/lib18f/WiFi/WiFiGeneric.cpp.o
*** [.pioenvs/energy-probe-appliance/src/main.cpp.o] Error 1
========================= [FAILED] Took 51.12 seconds =========================
```

RobTillaart, 2023-08-09:
Thanks for reporting this issue. Unfortunately I have no experience with ESPhome.
Will look at your logs later today if time permits to see if I can help.

chris-kuhr, 2023-08-09:
Here is a repo that utilizes this one and is included with a wrapper:
https://github.com/marianomd/acs712-esphome.git

RobTillaart, 2023-08-09:
```
/config/.esphome/platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal.h:70:16: error: macro "yield" passed 1 arguments, but takes just 0
 void yield(void);
```

This is an ESP32 error, **yield()** is not used in the ACS712 version 0.3.7, so the library does not cause this.

----

The other errors look like linking errors, they refer to "timing" functions and a mismatch in definitions in esphome::
```
millis(), micros(), delay(), delayMicroseconds()
```

There is a problem of redefinition with includes:
- src/esphome/core/hal.h
- /config/.esphome/platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal.h

So I expect one of them should be removed (or patched), or ???
Disclaimer:  no ESPhome experience

Q: if you remove the ACS712 library from the project, are all errors gone?

----

So, as far as I can tell there is no problem pointing to the ACS712 library.
All errors seem to be ESP32 / ESPhome related.

RobTillaart, 2023-08-09:
> Here is a repo that utilizes this one and is included with a wrapper: https://github.com/marianomd/acs712-esphome.git

Think I understand that code, and it seems nowhere related to the errors you see.

chris-kuhr, 2023-08-09:
Thanks for the clarification. 
And yes, without it, it compiles just fine. Here the output:

```INFO ESPHome 2023.7.1
INFO Reading configuration /config/energy-probe.yaml...
WARNING GPIO12 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Compiling app...
Processing energy-probe-appliance (board: esp32-poe-iso; framework: arduino; platform: platformio/espressif32@5.3.0)
--------------------------------------------------------------------------------
Removing unused dependencies...
Library Manager: Removing ACS712 @ 0.3.7+sha.c56d3ce
INFO Removing ACS712 @ 0.3.7+sha.c56d3ce
Library Manager: ACS712@0.3.7+sha.c56d3ce has been removed!
INFO ACS712@0.3.7+sha.c56d3ce has been removed!
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- WiFi @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- Update @ 2.0.0
|-- noise-c @ 0.1.4
Compiling .pioenvs/energy-probe-appliance/src/esphome/components/adc/adc_sensor.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/components/api/api_connection.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/components/api/api_frame_helper.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/components/api/api_pb2.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/components/api/api_pb2_service.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/components/api/api_server.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/components/api/list_entities.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/components/api/proto.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/components/api/subscribe_state.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/components/api/user_services.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/components/esp32/core.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/components/esp32/gpio.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/components/esp32/preferences.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/components/ethernet/esp_eth_phy_jl1101.c.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/components/ethernet/ethernet_component.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/components/md5/md5.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/components/mdns/mdns_component.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/components/mdns/mdns_esp32.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/components/mdns/mdns_esp8266.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/components/mdns/mdns_host.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/components/mdns/mdns_rp2040.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/components/network/util.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/components/ota/ota_backend_arduino_esp32.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/components/ota/ota_backend_arduino_esp8266.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/components/ota/ota_backend_arduino_rp2040.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/components/ota/ota_backend_esp_idf.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/components/ota/ota_component.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/components/sensor/automation.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/components/sensor/filter.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/components/sensor/sensor.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/components/socket/bsd_sockets_impl.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/components/socket/lwip_raw_tcp_impl.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/components/socket/socket.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/core/application.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/core/color.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/core/component.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/core/component_iterator.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/core/controller.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/core/entity_base.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/core/helpers.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/core/log.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/core/scheduler.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/core/string_ref.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/core/time.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/esphome/core/util.cpp.o
Compiling .pioenvs/energy-probe-appliance/src/main.cpp.o
[... 110 lines cut ...]
Compiling .pioenvs/energy-probe-appliance/FrameworkArduino/esp32-hal-sigmadelta.c.o
Compiling .pioenvs/energy-probe-appliance/FrameworkArduino/esp32-hal-spi.c.o
Compiling .pioenvs/energy-probe-appliance/FrameworkArduino/esp32-hal-time.c.o
Compiling .pioenvs/energy-probe-appliance/FrameworkArduino/esp32-hal-timer.c.o
Compiling .pioenvs/energy-probe-appliance/FrameworkArduino/esp32-hal-tinyusb.c.o
Compiling .pioenvs/energy-probe-appliance/FrameworkArduino/esp32-hal-touch.c.o
[... cut ...]
Flash: [===       ]  33.5% (used 613829 bytes from 1835008 bytes)
Building .pioenvs/energy-probe-appliance/firmware.bin
Creating esp32 image...
Successfully created esp32 image.
esp32_create_combined_bin([".pioenvs/energy-probe-appliance/firmware.bin"], [".pioenvs/energy-probe-appliance/firmware.elf"])
Wrote 0xa6300 bytes to file /config/.esphome/build/energy-probe-appliance/.pioenvs/energy-probe-appliance/firmware-factory.bin, ready to flash to offset 0x0
========================= [SUCCESS] Took 87.96 seconds =========================
```

RobTillaart, 2023-09-05:
@chris-kuhr 
Any progress made with this issue?
Can it be closed (as it is not this library related) ?

chris-kuhr, 2023-09-05:
not yet, the author of the library wanted to look at it, when he finds the time. no news since...

RobTillaart, 2023-11-05:
As this issue is open for two months with no progress I close it.
Feel free to reopen when new information is available

closed 2023-11-05 by RobTillaart