RobTillaart/ACS712 #42: ACS712  sensor is reading the wrong value(current)

yvanddniyo, 2023-10-15:
When we connect the load in our system LCD display different value far or lower from the load current value, plus when you connect you see current displaying in LCD with no-load,  anyone has an answer or any kind of help?

RobTillaart, 2023-10-15:
(This is more a problem for the Arduino Forum I think).

I have not enough information to analyze the problem, so I can guess:

- does the load has the same power source as the LCD?
- is it capable of delivering enough amperes?
- can you compare the values on screen with a DMM (digital multi meter).
- if you disconnect the LCD and print the values to Serial instead do you have the same problem?

RobTillaart, 2023-11-05:
As there is no information I close this issue.

closed 2023-11-05 by RobTillaart