RobTillaart/ACS712 #44: Is there a similar implementation for ACS725

kgalilio, 2023-11-11:
The ACS725 is a 3.3 V sensor. It does not require a voltage divider if we increase the ADC gain. For the 5A bidirectional [xLLCTR-05AB](https://www.allegromicro.com/-/media/files/datasheets/acs725-datasheet.pdf), the resolution is 264mV/A, and the zero current point = Vcc x 0.5.
Is there any way to implement this sensor in your library?

RobTillaart, 2023-11-11:
Thanks for your question,
Need to check the datasheet, and maybe it can be a derived class. Otherwise a seperate class/lib can be based upon same /similar interface.
As I do not have a device I cannot do tests 

(Pretty busy so I will investigate coming week)

RobTillaart, 2023-11-11:
At first look it looks pretty compatible
Have you tried the following constructor?
```cpp
ACS712  mySensor(A0, 3.3, 1023, 264);
```

A0 is analog pin of the MCU.

, so I can create a derived class so it looks nice...

kgalilio, 2023-11-11:
Hi Rob,

Thanks for the update and for your help.
I am using esp32 mcu with the esphome flavor of this library and, yes, I
have tried mySensor(34, 3.3, 4095, 264);
It is working, but the values are a little off. I suspect it has something
to do with the zero amps offset (Vcc x 0.5).
I have the 5A bidirectional version of the ACS725 supplied directly from a
3.3V power source on the PCB.

When there is no load, my current clamp sensor reads 1mA. The ACS725 reads
between 8mA and 13mA.

The electricity in my area is 230V.
I have a 60W bulb connected to the sensor. When the light is switched on,
the clamp sensor reads 231mA, and the ACS725 sensor reads 285mA.

Connecting a 75W bulb:
  Clamp: 285mA
  ACS725: 359mA

I hope this helps.

Cheers,

Khalil

kgalilio, 2023-11-11:
``

> At first look it looks pretty compatible Have you tried the following constructor?
> 
> ```c++
> ACS712  mySensor(A0, 3.3, 1023, 264);
> ```
> 
> A0 is analog pin of the MCU.
> 
> I can create a derived class so it looks nice...

I am using esp32 mcu with the esphome flavor of this library.
yes, I have tried `ACS712 mySensor(34, 3.3, 4095, 264);`
It is working, but the values are a little off. I suspect it has something to do with the zero amps offset (Vcc x 0.5).

I have the 5A bidirectional version of the ACS725 supplied directly from a 3.3V power source on the PCB.
The electricity in my area is 230V.
Using a 60W bulb connected to the sensor, a current clamp sensor reads 231mA, and the ACS725 sensor reads 285mA.

Using a 75W bulb:
  Clamp: 285mA
  ACS725: 359mA

I hope this helps.

RobTillaart, 2023-11-11:
The library has functions to calibrate the midpoint, including an auto-midpoint function. That should improve the zero amp level.

If that is reasonably stable, you can do the second calibration. The number 264, should be smaller how much can be calculated but can be done too with a few trial runs. I would start with 240 or so.

Adjusting this number seems strange but you correct for deviations in the ADC etc.

kgalilio, 2023-11-11:
Reducing the mV/A from 264 to 240 made it worse. I had to increase it to 320 to get a closer value to my Current clamp sensor's readings. Maybe this has something to do with the adc attenuation setting. Using the [example esphome code](https://github.com/marianomd/acs712-esphome) does not include settings for attenuation. This caps the esp32 adc pin to 1.1V; while the ACS725 5A sensor should go up to 2.97 with 264mV/A. Maybe if we cater for the attenuation: 11dB, the readings will be more accurate.

RobTillaart, 2023-11-12:
OK if 320 works better than 240 I probably made a math mistake in my head (not uncommon).
Good part is that you got better results.

I'm not familiar with the esphome framework, but if the ESP32 ADC is set to only use a range from 0 to 1.1V 
than you should probably call
```cpp
ACS712 mySensor(34, 1.1, 4095, 264);       //  264 to be tuned 
```

RobTillaart, 2023-11-23:
@kgalilio 
Any results to report?

RobTillaart, 2023-11-29:
@kgalilio 
As no results are reported or additional questions are asked for some weeks, I close the issue.
Feel free to reopen if needed

kgalilio, 2023-11-30:
> OK if 320 works better than 240 I probably made a math mistake in my head (not uncommon). Good news is that you got better results.
> 
> I'm not familiar with the esphome framework, but if the ESP32 ADC is set to only use a range from 0 to 1.1V than you should probably call
> 
> ```c++
> ACS712 mySensor(34, 1.1, 4095, 264);       //  264 to be tuned 
> ```

@RobTillaart: My apologies, I had to shift all my IT gear to a new location and hardly had the time to check my personal mail.
I tried the suggested values, but the results are inconsistent.

**I tried 195W load with the following:**
`ACS712 *ACS = new ACS712(34, 2.9, 4095, 244);`
_The Amps clamp sensor reads .795 A and the ACS725 fluctuate around .791 A (.791 ~ .805)_

**120W load:**
_The Amps clamp sensor reads .483 A and the ACS725 fluctuate around .498 A (.498 ~ .507)_

**60W load:**
_The Amps clamp sensor reads .229 A and the ACS725 fluctuate around .240 A (.238 ~ .252)_

**195W load:**
`ACS712 *ACS = new ACS712(34, 1.1, 4095, 83);`
_The Amps clamp sensor reads .795 A and the ACS725 fluctuate around .795 A (.787 ~ .800)_

**120W load:**
_The Amps clamp sensor reads .483 A and the ACS725 fluctuate around .443 A (.435 ~ .446)_

**60W load:**
_The Amps clamp sensor reads .229 A and the ACS725 fluctuate around 0 A (0 ~ .017)_

I am not sure where to go from here!

[image]

RobTillaart, 2023-11-30:
What current do you measure with no load?

kgalilio, 2023-12-01:
> Start at the begin, what do we measure. 195Watt / .795A = 245volt 120 / .483 =248 volt 60 /.229 = 263 volt
> 
> The assumption is that your wattage is correct, however the 3 measurements lead to different voltage. So can you check the voltage of the load?
> 
> Can you measure to the voltage of the sensor. Given the results it should be near the 2.9 (as first measurement seems closer to the reality.
> 
> Please provide at least 3 digits
> 
> Note that you give an important parameter voltage with only 2 significant digits, so you may not expect more than 2 correct digits in the answer.

The precise DC voltage of the sensor is 3.306V
The Voltage of the AC load measured by the sensor is 236.5V

kgalilio, 2023-12-01:
> What current do you measure with no load?
> 
> With the given constructors.

With 0 load, my clamp meter shows .001 ~ .002
The ACS725 shows the following:
(34, 2.9, 4095, 244) = .003 ~ .007 (I have not changed the noise from the default value 43)
(34, 1.1, 4095, 83) = 0.0

RobTillaart, 2023-12-01:
> I really appreciate your help.

You're welcome

The ESP32 has non-linearities in its ADC in the upper / lower end. 
Check e.g. -https://w4krl.com/esp32-analog-to-digital-conversion-accuracy/

_The ESP32 is a powerful microcontroller with many input/output ports. Specifically, it contains two 12-bit multiplexed analog to digital converters (ADCs) for a total of 18 channels. ADC1 is attached to eight GPIOs from 32 to 39. ADC2 is attached to 10 GPIOs (0, 2, 4, 12 to 15 and 25 to 27). There are some restrictions on the use of ADC2 when the WiFi transceiver is active. Since most applications are WiFi-enabled, it is best to use ADC1 when possible._

So 12 bit = 0 - 4095 not 8191 (13 bit is quite unusual, although?)
Q: which version of the ESP32 do you have S1, S2 or S3?  they might differ from the link above?

---- 

> I used a multimeter to measure the voltage from the ACS725 going into pin
34 of my ESP32, and the value is 1.653V.

That is perfect, do you use **autoMidPoint()** in your code?

kgalilio, 2023-12-07:
> @kgalilio <https://github.com/kgalilio>
>
> Any progress to report?
> Can the issue be closed now?

Hi Rob,
I think the issue is to do with the ESP32 ADC capabilities.
I ended up taking the ACS712 readings and applying a polynomial calibration
to it. So far so good.
Please go ahead and close the issue.
Thanks for your help.

RobTillaart, 2023-12-07:
@kgalilio 

You might consider using my multimap library for the non linear correction. 
Might be good enough...