RobTillaart/ACS712 #50: interfacing acs712 with esp32

vishaala, 2024-03-19:
hii, I want to interface acs712(5 A) with ESP32(Doit esp32 Devkit v1) and measure AC RMS current through a load. I am confused about the isuue #15 . Whether  do i need to connect the voltage divider or not,  if needed, how to connect and the resistor values??Where do i connect the sensor output (like which pin , because some pins cant be interfaced if we use wifi as I read). Can you please tell me the hardware connections and corresponding arduino IDE code for it from scratch(step by step). I request you to solve my confusions.

vishaala, 2024-03-19:
here are some extra details,  in my country  the voltage is 230V 50Hz AC.

RobTillaart, 2024-03-19:
Thanks for your question,
As I am quite busy this week it will take several days before having time to dive into your question. 
Sorry for the inconvenience.

Probably you will get answers faster on the Arduino forum. 
Have you searched youtube for ESP32 tutorials?
Expect there are several that explain it.

vishaala, 2024-03-19:
yeah, there are no youtube tutorils as far i searched, everything is for arduino . Thanks for your response . I request you to give  us (esp32 community) the solution as quick as possible.

vishaala, 2024-03-19:
I cannot find anything clear and crisp on the arduino forum(or rather anywhere) . There are many confusions related to this. I request you to give the exact hardware connections and code by taking your time. It will be very useful for my project and for others in future too.Thank you.

RobTillaart, 2024-03-20:
To get started I propose to use a voltage divider of a 20 KΩ and a 10 KΩ resistors (1% or better).
This will reduce the maximum voltage of the ACS with one third from 5.0 Volt to 3.3 Volt.
This will reduce the signal to be within range of the ESP32.

```
ACS712-OUT ----- 10 KΩ -----X------20 KΩ ------- GND
                            |
                            |
                            |
                        ADC of ESP32
```

The voltage divider will also reduce the 185 mV per Ampere steps with one third to 123.4 mV per Ampere.
So use 123.4 as the **mVperAmpere** parameter in the constructor.

```cpp
ACS712  ACS(34, 3.3, 4095, 123.4);
```

----

optionally you can use a voltage divider with two 10 KΩ resistors. That will reduce with one half.
So the parameter should become 92.5 mVperAmpere

Success

vishaala, 2024-03-20:
I request you to give the exact code for my purpose ( as there are some confusions in the void setup() function related to the setADC, Automidpoit functions...) And void loop() too.

RobTillaart, 2024-03-20:
See examples

vishaala, 2024-03-21:
[image]

Which example should I use for my purpose? Suggest one thing from those which gives accurate values of AC rms current

RobTillaart, 2024-03-21:
ACS712_20_AC_compare.ino

RobTillaart, 2024-03-21:
That compares both AC methods, and you can verify which gives the best values.
Be aware that the form factor of the AC will differ if there is noise on your 230 lines or when dimmers are used.
You can compensate for that with the appropriate function. Default the library uses sqrt(2)/2 ~~ 0.707

vishaala, 2024-03-21:
I will test and give the result so if I have problems I will report.

vishaala, 2024-03-23:
I have done the testing, the current value always shows greater than the original value when the noise value is higher. How can I reduce the noise?

vishaala, 2024-03-23:
My circuit
[image]

vishaala, 2024-03-23:
My code:

//
//    FILE: ACS712_20_AC_compare.ino
//  AUTHOR: Rob Tillaart
// PURPOSE: demo AC measurement comparison
//     URL: https://github.com/RobTillaart/ACS712

#include "ACS712.h"

//  Arduino UNO has 5.0 volt with a max ADC value of 1023 steps
//  ACS712 5A  uses 185 mV per A
//  ACS712 20A uses 100 mV per A
//  ACS712 30A uses  66 mV per A

//ACS712  ACS(A0, 5.0, 1023, 100);
//  ESP 32 example (might requires resistors to step down the logic voltage)
ACS712  ACS(35, 3.3, 4095, 123.4);

void setup()
{
  Serial.begin(115200);
  while (!Serial);
  Serial.println(__FILE__);
  Serial.print("ACS712_LIB_VERSION: ");
  Serial.println(ACS712_LIB_VERSION);

  ACS.autoMidPoint(50,100);
  //ACS.setMidPoint(1903);
  Serial.print("MidPoint: ");
  Serial.print(ACS.getMidPoint());
  

  float a = ACS.mVNoiseLevel(50,100);
  ACS.setNoisemV(a); //25mV if needed or a if
  Serial.print(". Noise mV: ");
  Serial.println(ACS.getNoisemV());
  ACS.suppressNoise(1);

}

void loop()
{ 
  
  //int m1 = ACS.mA_AC(50,60);
  int m2 = ACS.mA_AC_sampling(50,90);
  //Serial.print("mA:\t");
  //Serial.print(m1);
  //Serial.print("\t\t"); 
  if(m2<400) m2=0;
  Serial.println(m2);
  //delay(1000);
}

//  -- END OF FILE --

vishaala, 2024-03-23:
Output 
When the load is off:
[image]

When the load is on:
[image]

While the load draws 650 to 690 milliAmps, this shows much  greater value. But the one thing I noticed is that when I reset the program and sometimes the noise gets reduced (because of some unknown factors) , the values are accurate. So I think the noise is the problem. What may be the unknown factors or any other things??

RobTillaart, 2024-03-23:
> I have done the testing, the current value always shows greater than the original value when the noise value is higher. How can I reduce the noise?

What did you exactly test?
What do you mean by original value?
What levels of noise do you measure?

Noise will always disturb peak to peak measurements as you dont know how the peak is affected. The sampling method has the stochastic behaviour (if enough measurements are made) that the + and - noise cancel each other out.

vishaala, 2024-03-23:
Please observe the details given and give me some inference, it's my kind request

RobTillaart, 2024-03-23:
Please do not post images of output.
Text can be used for analysis (spreadsheet).
Images cannot.

vishaala, 2024-03-23:
Sorry

RobTillaart, 2024-03-23:
> What may be the unknown factors or any other things??

Loose wires?
Unshielded wires?
Running motors nearby?
Indecent power supplies?
Noise in the AC signal itself?
Fake ACS712 chip?
....

vishaala, 2024-03-23:
I have used jumper wires and breadboard

RobTillaart, 2024-03-23:
E.g. 
if the power supply of the ESP32 is not stable, it might affect the ADC readings.

vishaala, 2024-03-23:
While load off:

12:33:12.130 -> D:\8thSemProject\ACS712_20_AC_compare\ACS712_20_AC_compare.ino
12:33:12.130 -> ACS712_LIB_VERSION: 0.3.9
12:33:16.110 -> MidPoint: 2094. Noise mV: 63
12:33:19.925 -> 0
12:33:21.736 -> 0
12:33:23.543 -> 0

vishaala, 2024-03-23:
While load on:

12:34:12.266 -> 0
12:34:14.093 -> 479
12:34:15.863 -> 808
12:34:17.665 -> 808
12:34:19.489 -> 806
12:34:21.305 -> 806
12:34:23.104 -> 807
12:34:24.891 -> 809

RobTillaart, 2024-03-23:
So zero point looks ok

RobTillaart, 2024-03-23:
So current is stable around 808.
What value did you expect?

vishaala, 2024-03-23:
> E.g. 
> if the power supply of the ESP32 is not stable, it might affect the ADC readings.
> 
> 

I have checked the value with multimeter it's ok like 5.05 volts

vishaala, 2024-03-23:
> So current is stable around 808.
> What value did you expect?

650 to 690 milliAnps, this is the max, it can't go greater than this

vishaala, 2024-03-23:
> So zero point looks ok

Yeah I think so,  noise is the thing that makes the noise, I guess

RobTillaart, 2024-03-23:
Note that the noise level often is in the mV level.
Your multimeter does not show 3rd or 4th decimal.
That said, lets assume it is stable.

vishaala, 2024-03-23:
> Note that the noise level often is in the mV level.
> Your multimeter does not show 3rd or 4th decimal.
> That said, lets assume it is stable.
> 
> 

Yeah you are right, but esp32 power supply is regulated I guess

RobTillaart, 2024-03-23:
So you read 808 and you expect ~670.
That is ~15% off, 

Be aware that with these values you use less than 
20 % of the ADC range == max 1000 steps.
So your precision will be in the order of 0.5% if all rest is perfect stable etc.

RobTillaart, 2024-03-23:
15% off is way more than normal noise.

Can you put your multimeter in series with your load so you can compare the current measurements with the multimeter?

vishaala, 2024-03-23:
> 15% off is way more than normal noise.
> 
> Can you put your multimeter in series with your load so you can compare the current measurements with the multimeter?

Yeah the multimeter measures ~670mA

vishaala, 2024-03-23:
[image]

The jumper wires do slightly shake because of the ceiling fans air , so do  they make any problem

vishaala, 2024-03-23:
> My code:
>
> //
> //    FILE: ACS712_20_AC_compare.ino
> //  AUTHOR: Rob Tillaart
> // PURPOSE: demo AC measurement comparison
> //     URL: https://github.com/RobTillaart/ACS712
>
>
> #include "ACS712.h"
>
>
> // [...]

Is there any problem in the code corresponding to the circuit

RobTillaart, 2024-03-23:
> Yeah the multimeter measures ~670mA

So sometimes the measurements are quite good as you had 650-690 mA. That means there is no error in the library or the code you use.

> The jumper wires do slightly shake because of the ceiling fans air, so do they make any problem

No, what you see is a higher current and it is rather stable. More current means a lower resistance. Loose wires give higher resistance.

A fan has a motor which can generate a decent amount of noise. Do you see differences when you switch off all fans?

RobTillaart, 2024-03-23:
Q: what is the load you use?

vishaala, 2024-03-23:
Resistive load (incandescent bulb)

RobTillaart, 2024-03-23:
> Is there any problem in the code corresponding to the circuit

No as you get values.

vishaala, 2024-03-23:
> So sometimes the measurements are quite good as you had 650-690 mA. That means there is no error in the library or the code you use.
>
>
> No, what you see is a higher current and it is rather stable. [...]

No change, but everytime I reset the esp32 by pressing reset,   the noise value keeps changing drastically around min 40mV to max 130mV , everytime I reset. There is something to do with the supply I guess

RobTillaart, 2024-03-23:
Can you add a delay(5000) in setup.
That could give everything time to stabilize.

vishaala, 2024-03-23:
Where should I exactly add the delay in the code

RobTillaart, 2024-03-23:
> the noise gets found automatically through the code, so the loads need to be off initially while measuring the noise. So how can it be a problem with the load or home supply.

The noise measurement is only done once in time.
The code assumes that the noise level is constant after that.

It might be interesting to run a sketch that measures the noise constantly and plot it. 
It might give hints what causes the extreme noise.

RobTillaart, 2024-03-23:
> Where should I exactly add the delay in the code

As first statement in setup()

vishaala, 2024-03-23:
> The noise measurement is only done once in time.
> The code assumes that the noise level is constant after that.
>
> It might be interesting to run a sketch that measures the noise constantly and plot [...]

Yeah I will do it

RobTillaart, 2024-04-04:
@vishaala 
Any progress to report?

RobTillaart, 2024-04-11:
As there was no answer I assume the problem is solved and I close the issue.

closed 2024-04-11 by RobTillaart