RobTillaart/ACS712 #51: Experience Sharing - ACS712 on ESP32 with ESP Home for DC Measurement

uTZFu, 2024-10-08:
Hey Rob. -thanks for your great work here - really, really cool! I tried to measure the power consumption for my setup for a whole day and finally made it :) Since I struggeled a lot and searched countless forums, i wanted to share my experience how I made it here (in case it is of help to other noobs like me): 

Setup: 
- Aim: Measuring power consumption of an interactive company sign.
- ESP32 Wroom Dev Board with multiple sensors (DHT, MIC, LD, Speaker, and multiple LEDs) integrated via ESP Home into Home Assistant.
- Power supply. 5v (Measured 5,18 V) with a maximum of 18 amps (unlikely to be ever drawn - standard would be up to 8 a)
- ACS712 with 20a threshold
- Wiring: ACS712 connected directly (without any voltage divider) to GPIO34
- Relevant Code used for the ESP32 within ESP Home:
`esphome:
  
  libraries:
    - https://github.com/RobTillaart/ACS712.git 
  includes:
    - acs712_component.h #Should be stored in your ESP Home Folder (directly - no sub folder needed)

esp32:
  board: esp32dev
  framework:
    type: arduino
    
sensor:
  - platform: adc #Displays the value measured at the GPIO of your ESP32 - for control purposes  - can be deleted if finished
    pin: GPIO34 
    name: "Stromsensor für Leistungsaufnahme"
    update_interval: 5s
    attenuation: auto #Essential part for the ESP32 since they are otherwise skipping out any voltage higher 1,01V - Alternative Calibration can be done

  - platform: custom
    lambda: |-
      auto acs712_sensor = new ACS712Sensor();
      App.register_component(acs712_sensor);
      return {acs712_sensor->current_sensor, acs712_sensor->power_sensor};
    sensors:
    - name: "Stromaufnahme in Ampere"
      unit_of_measurement: A
      accuracy_decimals: 2
    - name: "Aktuelle Leistung in Watt"
      unit_of_measurement: W
      accuracy_decimals: 2
`
Within the File ACS712.h that I shared by @Marianomd I had to do the following minor adjustments

`#include "ACS712.h"

class ACS712Sensor : public PollingComponent {
   public:
    ACS712 *ACS = new ACS712(34, 3.3, 4095, 100); // Connected to GPIO34, 3.3 Volts being the maximum Voltage on an ESP32, 4095 being the maximum steps for 12bit (4096 is the number of steps, while 4095 is the number of options available and 100 being the factor for my 20a Version fo the AC712 without any Voltage divider, so no recalculation needed)
    Sensor *current_sensor = new Sensor();
    Sensor *power_sensor = new Sensor();

    ACS712Sensor() : PollingComponent(10000) {} // Changed Polling interval to 10 Seconds instead of 15 to get a faster response

    void setup() override {
        ACS->autoMidPoint();
        ESP_LOGD("acs712", "MidPoint: %d", ACS->getMidPoint());
        ACS->setNoisemV(21);
        ESP_LOGD("acs712", "Noise mV: %d", ACS->getNoisemV());
    }

    void update() override {
        float average = 0;
        int count = 5;
        for (int i = 0; i < count; i++) {
            average += abs(ACS->mA_DC()); // Two changes here: For an AC Measurement it should be mA_AC(), since I only measure DC, i changed it accordingly. the second add on was the abs() added, since I did not wanted to rewire the load but had negative values for a and p :) - Works perfectly fine for me
        }
        float amps = average / count / 1000.0;
     
        current_sensor->publish_state(amps);
        power_sensor->publish_state(amps * 5.18); // Since my power supply delivers 5.18 V and not 220 or 230 as in DC, this is required to get the correct Power consumption
    }
};`

Hope this is the correct way to share this and that it helps someone with as little experience as me :-)

RobTillaart, 2024-10-09:
Thanks for this piece of documentation.
Think I wrap it in a separate file to add to the repo so it is easier to find (maybe).

RobTillaart, 2024-10-20:
@uTZFu 
Added link to this issue in the readme.md, 
I will close this issue, but feel free to add new comments or insights.

Again, thanks for sharing your experiences, appreciated

closed 2024-10-20 by RobTillaart

ad-on-is, 2024-11-02:
Can you guys help me out please? I don't get any readings when I connect A0 to the OUT of the ACS712.

```
19:30:40	[D]	[sensor:093]	'Amperes': Sending state 0.00000 A with 5 decimals of accuracy
19:30:40	[D]	[sensor:093]	'Watts': Sending state 0.00000 W with 5 decimals of accuracy
19:30:41	[D]	[sensor:093]	'Amperes': Sending state 0.00000 A with 5 decimals of accuracy

```
However, when the A0 is floating (not connected) I do get some readings, so the script seems to work fine (I think). 
I tried with resistors, and without, and the result is the same. 

Also, I only need to check whether current is flowing or not, I don't really need exact measurements.

RobTillaart, 2024-11-03:
@ad-on-is 

As this is not a problem of the library (you get readings) I propose to ask this question on the Arduino forum or ESP (home) forum. I do not have any ESP home hands-on experience.

Only cause I can think of are bad connections, shortcut with GND maybe? There is too little information to draw any (other) conclusion so far.

uTZFu, 2024-11-03:
@ad-on-is: It took me quite some time to set the ACS712 up with esphome like described above and I am still on the steep part of my learning curve wrt to this stuff. What helped me was: During setup i did read the analoge voltage values at GPIO34 to see if the changes are somehow plausible if I increase / decrease power on the system. What does your GPIO deliever if you change power consumption? Does it change accordingly and in what boundaries?