RobTillaart/ACS712 #53: Using acs712 with esp32

Smatris, 2024-12-13:
Using the example you created (examples/ACS712_20_AC/ACS712_20_AC.ino) as a basis, using your library, what would change if I were to use it to measure current in the ESP32? In addition to the analog inputs?

This is my code, but the output on the serial monitor is 0, it does not measure the current

#include <ACS712.h>
#include <ZMPT101B.h>

#define ACS712_PIN 36          // Pino analógico para o sensor de corrente
#define ACS712_MVPERAMP 185.0  // Sensibilidade do sensor ACS712 5A (mV/A)

// Configuração do sensor ZMPT101B (tensão)
#define ZMPT101B_PIN 34 
#define ZMPT101B_SENSITIVITY 500.0f // Sensibilidade padrão do ZMPT101B
#define ZMPT101B_FREQUENCY 60.0 

// Instâncias dos sensores
ACS712 currentSensor(ACS712_PIN, 3.3, 4095, ACS712_MVPERAMP); // Configura o sensor ACS712
ZMPT101B voltageSensor(ZMPT101B_PIN, ZMPT101B_FREQUENCY);     // Configura o sensor ZMPT101B

void setup() {
  Serial.begin(9600);
  delay(2000); // Aguarde o monitor serial abrir

  // Configuração do sensor de corrente
  Serial.println("Inicializando o sensor de corrente...");
  currentSensor.autoMidPoint(); // Auto-calibração do ponto médio
  Serial.println("Sensor de corrente inicializado!");

  // Configuração do sensor de tensão
  Serial.println("Inicializando o sensor de tensão...");
  voltageSensor.setSensitivity(ZMPT101B_SENSITIVITY);
  Serial.println("Sensor de tensão inicializado!");
}

void loop() {
  // Medição de corrente
  float current = currentSensor.mA_DC(10) / 1000.0; // Mede corrente DC (A)
  Serial.print("Corrente (DC) = ");
  Serial.print(current);
  Serial.println(" A");

  // Depuração da corrente
  int rawCurrent = analogRead(ACS712_PIN);
  Serial.print("Leitura bruta (ACS712): ");
  Serial.println(rawCurrent);

  // Medição de tensão
  float voltage = voltageSensor.getRmsVoltage(10); // Mede tensão RMS
  Serial.print("Tensão RMS = ");
  Serial.print(voltage);
  Serial.println(" V");

  // Depuração da tensão
  int rawVoltage = analogRead(ZMPT101B_PIN);
  Serial.print("Leitura bruta (ZMPT101B): ");
  Serial.println(rawVoltage);

  // Espera 1 segundo antes de nova leitura
  delay(1000);
}

RobTillaart, 2024-12-14:
Thanks for your question.

You could try if just reading the analog port works. 

Due to personal circumstances it might take weeks before I can dive into this.
If you need a faster answer, please open a question on the Arduino Forum.

RobTillaart, 2024-12-30:
@Smatris 
Did you make any progress or does the issue still exist?
I am limited online again

RobTillaart, 2025-01-02:
@Smatris 

At least you should initialize the I2C and the INA228 properly

Give this a try.
```cpp
void setup() {
  Serial.begin(9600);
  delay(2000); // Aguarde o monitor serial abrir

  Wire.begin();
  if (!currentSensor.begin() ) 
  { 
    Serial.println("Initialization current sensor failed");
  }

  // Configuração do sensor de corrente
  Serial.println("Inicializando o sensor de corrente...");
  currentSensor.autoMidPoint(); // Auto-calibração do ponto médio
  Serial.println("Sensor de corrente inicializado!");

  // Configuração do sensor de tensão
  Serial.println("Inicializando o sensor de tensão...");
  voltageSensor.setSensitivity(ZMPT101B_SENSITIVITY);
  Serial.println("Sensor de tensão inicializado!");
}
```

RobTillaart, 2025-01-11:
@Smatris 

Does the problem still exist? if not I close the issue.

RobTillaart, 2025-01-14:
@Smatris 
Feel free to reopen if needed

closed 2025-01-14 by RobTillaart