RobTillaart/ACS712 #55: ESPHome component collabration

ChadMatsalla, 2025-03-01:
Rob, would you like to collaborate with me to produce an actual esphome component?

There is at least one [acs712_exte](https://github.com/grego1981/acs712_exte) uses your library as a platformio include. The ESPHome team to not prefer to do things this way, especially when using things like i2c. His esphome component also doesn't allow the user to specify the pin or update interval.

I forked his component and added that functionality, but the esphome maintainers suggested that if we could remove the library dependency it could be added to the esphome core of components. The path to that is as a PR to ESPHome. We could share credit for this, of course.

I want to add the capbility to use an I2C ADC like ADS1115 as esphome components.

I would try to copy ACS712.cpp and ACS712.h as they are to make this as easy as possible. I think it's do-able, and I'm doing it anyway in my private repo as a proof of concept.

Thoughts?

Chad

RobTillaart, 2025-03-01:
Mentioned this issue in #54 as there is a strong relation.

With respect to your question,

I do not have ESPhome experience other than a few quick reads, so I do not know if I can help. I have to go through a learning curve, probably make test setups etc which all takes time. And time is my scarcest resource, so for now I need to think if this is a project for me. to be continued.

Q1 : Why are you not doing this project in a public repo? Far more people could help.

Q2: Are there examples of I2C devices in ESPHome that have similar complexity and are supported? 
Links to such repo's could be useful to understand how the interface of an cpp library maps unto the ESPHome ecosystem.

RobTillaart, 2025-03-01:
Found a guide to transform a sketch to a component.
https://github.com/DirkHeinke/arduinoToESPHome-guide

The example is INPUT only so it does not show how to capture the output of a function, still informative.

RobTillaart, 2025-03-01:
Spent about 1.5 hr and I got an ESPHOME dashboard on my laptop. 
Can you post the config file you use, might be easier for me

ChadMatsalla, 2025-03-01:
Hi Rob,

Here's [my repo](https://github.com/ChadMatsalla/esphome_components) I recommend installing the esphome cli as found here: [Installing ESPHome manually](https://esphome.io/guides/installing_esphome.html).

They use venv. I'm a conda guy, so I do it this way:
```
$ conda create esphome
$ conda activate esphome
$ conda install python
$ pip3 install esphome
$ git clone https://github.com/ChadMatsalla/esphome_components.git
$ cd esphome_components/configs
$ esphome run acs712.yml
```
That will compile it. You'll see it pulling your library.

In that repo, you will see a `components/acs712` folder. If you look carefully there it contains the mechanism to allow esphome to use your library.

I propose that you be responsible for the contents of your library. I'll wrap the ESPHome bits like `sensor.py` around it.

I'll take care of interfacing with the ads1115.

Basically I'm asking your permission to take your library (giving you credit) and making it useful in ESPHome with an ADS1115 (or other ADC). Could that be a good start?

ChadMatsalla, 2025-03-01:
As for what one can do with ESPHome, it's a very very very fast way to define the components in a system in yaml and have esphome generate firmware. If you look in my `acs712.yml` config, you'll see a number of things:

I tell esphome that:
- I'm using your library
- I'm using I2C and on which pins
- I'm using a `seeed_xiao_esp32c3`
- I have wifi
- I'm publishing stuff from myself with esphomeapi
- I have an ADS1115 that's on the i2c bus described above
- That I have a ADC712 sensor

Because I haven't changed the code to use the ADS1115 (yet) I've specified `pin:` in the adc712 part.

If you have homeassistant, this esp32 will quickly join and that sensor will become part of your homeassistant ecosystem.

In the old days when I manually published mqtt this would have taken a very long time.

Finally, I added the `web_server:` so that you don't even need homeassistant - You can connect directly to the esp.

As for the dashboard, I don't use it. I do everything on the commandline, as I showed above. `run` compiles and flashes, `logs` connects over the network to a node.

Once you flash an esphome node and if you include the `ota:` directive, future flashes are done OTA.

ChadMatsalla, 2025-03-01:
ESPHome has a thing called `host:`. I use it for testing, this is why there's a config called `acs712-esp.yml` and `acs712-host-yml`. If you need help live I'm often found on the esphome discord (@dieselwurks). They are really good there. Just remember not to paste code, yaml, or images in the main channels - create a thread first. Ping me there at any time.

RobTillaart, 2025-03-01:
> I propose that you be responsible for the contents of your library.

No problem ;)

> Basically I'm asking your permission to take your library (giving you credit) and making it useful in ESPHome with an ADS1115 (or other ADC). Could that be a good start?

Sounds like a good start. You have permission (as by the MIT license) to use it for the ESPHome. It is not a problem for me to add an esphome folder in the ACS712 repo which adds the yaml files and how to etc when things are stable. Or at least a section in the readme.md file of the repo with a link to your repo.  Same for the ADS1x15 library.

Quite busy now, will "study" your repo in some more detail coming week to try to understand.

One question in front, how do you guard breaking changes? 
If I change the API for some reason (good bad or ugly), it should not break the ESPhome implementations. So somehow  the ESPHome wrapper should use a released version of the library, preferable latest of course.

RobTillaart, 2025-03-26:
@ChadMatsalla 

Did you make progress?
Does this issue need to be open? otherwise it can be closed.
If needed you can still reopen it,

brianmcentire, 2025-03-31:
Just want to say I am watching this with enthusiasm as I am currently working on a DIY project that this will be perfect for. Its a ESP32 based device, it has one relay already working that controls a 24v ac load. Because of the nature of the load, there are other relays on the circuit that are not part of this controller, so I also want to be able to detect when the circuit is energized (when my relay is closed, to test of the other(s) are as well). 

ESPhome makes this stuff so easy to access. Control of the relay based on other sensors via Home Assistant automation, as well as graphing of states built in, and the device is accessible over local wifi.

I was looking for ACS712 in Esphome and this looks like it is really close. I'm using someone else's fork of this repo (marianomd) and getting values, although not correct ones yet.

It will be great if this can end up as an official ESPHome component.

ChadMatsalla, 2025-03-31:
Hi all,

I was working on a native component for esphome - one that doesn't use this library at all as that's the method preferred by the esphome project. This permits the external component to use native esphome adc libraries, for example. Sadly this has fallen off of my to-do list and I'm not sure when it will return.

My use case was to detect that a non-zero amount of current was flowing to a solenoid. I've found that it's safer and "good-enough" to use a CT-clamp based solution for my problem. That's why I've stopped on an external component.

Rob, this is good work!

closed 2025-03-31 by ChadMatsalla

RobTillaart, 2025-03-31:
@ChadMatsalla 

Thanks for the update!