RobTillaart/ACS712 #56: no matching function for call to 'ACS712::ACS712(int, double, int, int)'

FarmallDude, 2025-04-16:
I copied the code from the examples, hit upload and get this error. I'm using an ESP32 WROOM-32, if that matters. I even tried it on an ArduinoUNO R4. Doesn't seem to matter which example I try. What am I missing?

//
//    FILE: ACS712_20_AC.ino
//  AUTHOR: Rob Tillaart
// PURPOSE: demo AC measurement with point to point
//     URL: https://github.com/RobTillaart/ACS712

#include "ACS712.h"

//  Arduino UNO has 5.0 volt with a max ADC value of 1023 steps
//  ACS712 5A  uses 185 mV per A
//  ACS712 20A uses 100 mV per A
//  ACS712 30A uses  66 mV per A

//  ACS712  ACS(A0, 5.0, 1023, 100);
//  ESP 32 example (might requires resistors to step down the logic voltage)
  ACS712  ACS(34, 3.3, 4095, 66);

uint32_t start, stop;

void setup()
{
  Serial.begin(115200);
  while (!Serial);
  Serial.println(__FILE__);
  Serial.print("ACS712_LIB_VERSION: ");
  Serial.println(ACS712_LIB_VERSION);

  //  use simulation
  ACS.setADC(signal, 5, 1024);

  ACS.autoMidPoint();
  Serial.print("MidPoint: ");
  Serial.print(ACS.getMidPoint());
  Serial.print(". Noise mV: ");
  Serial.println(ACS.getNoisemV());
}

void loop()
{
  delay(100);
  start = micros();
  //  int mA = ACS.mA_AC();
  int mA = ACS.mA_AC_sampling();
  stop = micros();
  Serial.print("mA: ");
  Serial.print(mA);
  Serial.print(". Form factor: ");
  Serial.print(ACS.getFormFactor());
  Serial.print("  time: ");
  Serial.println(stop - start);
  delay(5000);
}

//  simulated 50 Hz signal
uint16_t signal(uint8_t p)
{
  return 512 + 400 * sin((micros() % 1000000) * (TWO_PI * 50 / 1e6));
}

 

TIA

RobTillaart, 2025-04-16:
@FarmallDude 

Tried to replicate the error message but could not. 
The reason is probably a different version of the ESP32 board environment.

The error message points toward the constructor, it looks like the default types of the constants mismatch.

To verify it is a datatype problem, could you change its line into

```
  ACS712  ACS((uint8_t)34, (float)3.3, (uint16_t)4095, (float)66);
```

RobTillaart, 2025-04-16:
Note for myself:
- 1st param could be int, 
- 2nd has to stay float ==> 3.3f in examples.
- 3rd has to stay uint16_t => int would exclude 16 bit ADC's 
- 4th could be made more explicit a float   66.0f

FarmallDude, 2025-04-17:
Hi Rob,
Thanks for the quick response!
I replaced the line but still get the error:
Compilation error: no matching function for call to
'ACS712::ACS712(uint8_t, float, uint16_t, float)'

I am using a 'cheap' board, from Amazon (
https://www.amazon.com/DORHEA-Development-Bluetooth-ESP-WROOM-32-ESP32-DevKitC-32/dp/B0B18JQF16/ref=sr_1_3?crid=2GKANE4AYLE9F&dib=eyJ2IjoiMSJ9.nR_nZfPt8cUgq_0NAqLeQRhEtXh3BgPHmgng8bcQi4OOhT4p9wasekx2c_XNRGor7Pbk1zuuVh9TowvCLI5GNd0Cvhppfs8A0ZXm1TFaqeo_lWDlY99CuJN2UrVv7XsbPuRKOOKdDtGzpaB9-vNr09W8BXlxU5Kpp2x6VNY2n6onDYLU7ulIla9kme_5hnzWx_RtkW3ascSgreK-VSe_M3UDYAHanPhUq6997ShhcCQGCH16SDtif14425joJFmk33-g7YQIVWl3lddI5yLVz1Rv7pQYOwuyzYUlBvPj8hw.Yx2I45qFlM0EK3bGubuY3tOjo2xBQZJABzJBYaDRIPk&dib_tag=se&keywords=3PCS%2BESP32%2BType%2BC%2BDevelopment%2BBoard%2BESP32S&qid=1744894258&s=electronics&sprefix=3pcs%2Besp32%2Btype%2Bc%2Bdevelopment%2Bboard%2Besp32s%2B%2Celectronics%2C102&sr=1-3&th=1
)
I have these boards installed in Arduino:
esp32 by Espressif Version 3.2.0
Arduino esp32 boards by Arduino Version 2.0.18

Attached is the code I'm using. As an overview, my project is attempting to
turn on a dust collector when a power tool is turned on by sensing the
current. Then, send an 'on' parameter via ESP_NOW to the receiver which
will power on the dust collector. But first, I need to get the power sense
piece to compile and work :)

I hope this helps, please let me know if you need/want any more information.

Cheers,
Bruce

RobTillaart, 2025-04-17:
> Compilation error: no matching function for call to
'ACS712::ACS712(uint8_t, float, uint16_t, float)'

Very strange as that is the exact constructor of the ACS712 library.
ACS712(**uint8_t** analogPin, **float** volts = 5.0, **uint16_t** maxADC = 1023, **float** mVperAmpere = 100)

Do you have more than one ACS712 library installed?
If not can you post the top 10 lines of the .h file, these should match this latest one.

```cpp
#pragma once
//
//    FILE: ACS712.h
//  AUTHOR: Rob Tillaart, Pete Thompson
// VERSION: 0.3.10
//    DATE: 2020-08-02
// PURPOSE: ACS712 library - current measurement
//     URL: https://github.com/RobTillaart/ACS712
//
// Tested with a RobotDyn ACS712 20A breakout + UNO.
//
```

> I am using a 'cheap' board, from Amazon 

The compilation does not depend on the hardware (should not) so gives no clue.

> esp32 by Espressif Version 3.2.0

I know this version has problems with I2C - see recent discussion https://github.com/RobTillaart/I2CKeyPad/issues/26 

Has nothing to do with this issue of the ACS712, just for you to be aware.

FarmallDude, 2025-04-17:
> RobTillaart/I2CKeyPad#26

I think there is only one, here is the whole file I found...

#ifndef ACS712_h
#define ACS712_h

#include <Arduino.h>

#define ADC_SCALE 1023.0
#define VREF 5.0
#define DEFAULT_FREQUENCY 50

enum ACS712_type {ACS712_05B, ACS712_20A, ACS712_30A};

class ACS712 {
public:
ACS712(ACS712_type type, uint8_t _pin);
int calibrate();
void setZeroPoint(int _zero);
void setSensitivity(float sens);
float getCurrentDC();
float getCurrentAC();
float getCurrentAC(uint16_t frequency);

private:
float zero = 512.0;
float sensitivity;
uint8_t pin;
};

#endif

Interestingly, in the IDE, if I try to downgrade to the previous version it
says "... already installed" but the "Install" button is there (instead of
"Remove" as I would expect). I'm going to try to manually remove the
library and re-install it. I'll let you know.

Cheers,
Bruce

FarmallDude, 2025-04-17:
OK, it seems re-installing the library solved that problem. Thanks for your
help!

Kind regards,
Bruce

RobTillaart, 2025-04-18:
Good to hear the problem is solved!

> I think there is only one, here is the whole file I found...

Comes from - https://github.com/shahid249/ACS712-Arduino-Library

closed 2025-04-18 by RobTillaart

FarmallDude, 2025-04-21:
Thank you Rob!